gpa-fryk-industries/CDIO #8: FPGA Signals

Atmelfan, 2018-10-13:
What signals are required between MCU and FPGA? Currently only basic configuration signals are connected (can be used as SPI once configuration is done).

What other signal should be shared? 
Suggestions:

- [ ]  Interrupt request (IRQ)
- [ ]  Master/slave control to avoid FPGA/MCU collision in case the FPGA can act as master

More suggestions?

davfr422, 2018-10-13:
There has to some kind of coherence for what's actually going on inside the FPGA during debugging. I guess SPI communication is an alright interface for sharing signals. Let's discuss this more in detail on monday. I feel like I need to better understand the issue in order to give a better answer.

Atmelfan, 2018-10-25:
Added ATT (ATTention) and MINH (Master INHibit) signals between the FPGA and MCU.

closed 2018-10-25 by Atmelfan